ArduPilot/ardupilot #17770: use arduino (Control Specialist)

RuffaloLavoisier, 2021-06-14:
## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

No :)

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

Arduino has a programming method for reading PWM signals. 
This allows the output device to be controlled on top of the Arduino.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

[image]

**Platform**
[ v ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

I have used this method as a personal project and I can actually solve the problem by connecting more various output devices.

khancyr, 2021-06-14:
hello

I fail to see where does this connect to ArduPilot ? You could always use a Y cable to get two PWM signal from your receiver : on for the flight controler and the other on to an arduino board.

RuffaloLavoisier, 2021-06-14:
The reason why I want to connect Arduino to the receiver is to allow the controller to program more features in Arduino using the value of 1000-2000 of the optional channel.

You can connect an extra channel to the receiver to the Arduino interrupt pin for use.

closed 2021-06-16 by IamPete1