ArduPilot/ardupilot #17780: OSDx_PLUSCODE_yyy not available in ardupilot

padcom, 2021-06-15:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**
When setting up OSD the PLUSCODE element is not available for planes. I have used it in copters though.

**Version**
4.1.0-dev from 2021-06-14

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying wing

**Hardware type**
Matek F405 WING

**Logs**
No logs needed

rmackay9, 2021-06-15:
I think it would be best to open a topic in the new Plane 4.1 category on discuss instead of raising an issue. https://discuss.ardupilot.org/c/arduplane/plane-4-1/160

I'll leave this open 'cuz I'm not the plane maintainer but normally that's how we do it I think.

IamPete1, 2021-08-08:
Plus-code is not enabled on the F405 wing to save flash. You can compile your own and change that here: https://github.com/ArduPilot/ardupilot/blob/a6e8f2e73bcf8fefd3b70e61346c5220d1395201/libraries/AP_OLC/AP_OLC.h#L24

Hopefully another thing the custom build server will fix, https://github.com/ArduPilot/ardupilot/issues/15179

closed 2021-08-08 by IamPete1