ArduPilot/ardupilot #17793: Copter: loiter "bucketing" when EKF origin is far from vehicle's location

rmackay9, 2021-06-17:
As part of the investigation of issue https://github.com/ArduPilot/ardupilot/issues/16736 we found that if the EKF origin is very far from the vehicle's actual location (say 60km or 120km) a copter flying in Loiter mode will only hold position at particular spots 50cm to 1m apart.

From the pilot's point of view they can fly the vehicle somewhat normally but when flying at low speeds in particular if the pilot pushes the sticks lightly the vehicle will move slowly but then when the sticks are release, the vehicle will return to its original position.  If the pilot pushes harder on the sticks, the vehicle will be able to push past a sort of virtual barrier and will re-settle on another point 50cm to 1m away.

tridge, 2021-06-18:
fix here: https://github.com/ArduPilot/ardupilot/pull/17803
works up to 800km

rmackay9, 2021-07-30:
This is resolved in master and the fix is also included in Copter-4.1.0-beta6.

closed 2021-07-30 by rmackay9