ArduPilot/ardupilot #17795: SET_POSITION_TARGET_GLOBAL_INT mavlink command not working at higher frequencies

07hokage, 2021-06-17:
## Bug report
SET_POSITION_TARGET_GLOBAL_INT command not being accepted by the copter at higher rates
**Issue details**
-> while sending the SET_POSITION_TARGET_GLOBAL_INT command using dronekit-python, if im sending the command  at 10Hz rate, it is not at all accepting the command. and when i close the script, it starts moving towards the target (assuming it is acting for the command sent before end of script ).  
-> as soon as the command is sent first time, i can see mission planner displaying  318>0 meaning the target has been received from the first time but copter couldn't respond before the second command is sent 
-> When i reduced the rate, it starts accepting commands, but when same target is being given in the loop, it starts acting in a wobble manner as if it is starting from rest every time it receives the command

While i had not faced this issue in 4.0.7( tested on actual hardware ) , but im facing this in 4.1.0-dev in SITL

**Version**
4.1.0-dev

**Platform**
 Copter

**Airframe type**
quad multirotor

**Hardware type**
SITL

**Logs**
[logs.zip](https://github.com/ArduPilot/ardupilot/files/6669510/logs.zip)

rmackay9, 2021-06-17:
Yes, this is a known issue with S-Curves.  PR https://github.com/ArduPilot/ardupilot/pull/17539 may resolve this.  By the way, if you're constantly sending in position targets at a very high rate, you might consider instead sending velocity targets... or slowing down the position target update rate.

07hokage, 2021-06-17:
Thanks @rmackay9  for the answer. Velocity commands works perfectly fine even at high rate.  But, is there anyway to access the **target position that the copter has stored** while executing the command? so that it will be easier to compare if my target is same as copter's i can skip sending again.

khancyr, 2021-06-17:
Hello,

You should be receiving https://mavlink.io/en/messages/common.html#POSITION_TARGET_GLOBAL_INT message indicating you what is the current target point

IamPete1, 2021-08-16:
sounds like this has been resolved.

closed 2021-08-16 by IamPete1