ArduPilot/ardupilot #17814: DJI MSP OSD Not Working On H7 Boards

MadRC, 2021-06-20:
In 4.1 Plane and Copter and daily builds the DJI MSP OSD feature is not working on H7 based boards.  

Zero OSD is coming up when configured correctly on these boards however It’s working fine on F4 boards such as the cube black.

Tested on 

Cube Orange 
765 Wing 

Not sure what’s driving the issues on these board types specifically. 

Multiple users now reporting the same issues and I have tested my self to confirm.

MadRC, 2021-06-20:
OK found a solution on Cube Orange, turning off HW Flow control ie for serial 1 setting BRD_SER1_RTSCTS =0 resolved it and OSD Works. 

However this does not work on the F765 Wing, interestingly there is only one option for this on serial 1 but it does not work. 
This is an F7 board.

ojoojoo, 2021-06-21:
Great news! I ordered an F405-WSE already because flying blind without OSD is basically not an option for me.
I will try the Pixhawk 4 mini tonight to see if your solution also works on this board.

Thanks a lot, and I really mean this, without people like you the hobby will grind to a hold eventually.

Have a great day!

Regards Orson

ojoojoo, 2021-06-22:
Goodmorning Ian,

I got the OSD working on the Pixhawk 4 mini, but only on serial1, because only on this UART have this option BRD_SER1_RTSCTS =0.
It’s not possible to make this work on other serial ports for now.

Have a great day.

Regards,
Orson

MadRC, 2021-06-22:
Interesting as that did not work on the F7 based 765.  I’m glad it did on the Mini though.

rmackay9, 2021-06-23:
By the way, I've added this issue to the [Copter-4.1 issues list](https://github.com/ArduPilot/ardupilot/issues/16478) so it won't be forgotten.  I briefly brought it up at this week's developer call but we didn't have time to discuss it in any depth.

MadRC, 2021-06-23:
Thanks @rmackay9 I did spot it.  If there is anything you guys want me to test it try please don’t hesitate to ask.  I don’t understand the MCU differences enough to be able to offer anything other than testing on this one sorry.

MadRC, 2021-06-23:
Ok I have got the Matek 765 wing working. 

It’s the same as the Cube Orange however you need to note that Serial 1 is actually UART 7 on the FC.  This caught me out when testing the other day. It’s working fine it’s just the UART layout is all over the place compared to the labelling as it’s a different setup.  

So again configure as normal, but as H7 and F7 turn off HW Flow Control as per above and connect DJI to TX and RX7 on the FC.

So for the 765 Wing. 

Configure Serial 1 = 33 
Set Baud = 115 
Set Osd_Type = 3 
Set BRD_SER1_RTSSCTS = 0

So basically to use the DJI system with F7 and H7 you need to turn off HW Flow Control, perhaps this simply now needs an update to the wiki rather than further investigation.

ojoojoo, 2021-06-23:
Great news, I will try it on mine out as soon as I can, thanks again for your efforts!!

yaapu, 2021-07-05:
@tridge can we disable RTS and CTS right at serial port init when instantiating the backend?

yaapu, 2021-07-08:
this https://github.com/ArduPilot/ardupilot/pull/17965 should allow us to set RTS and CTS at runtime for the MSP backends

rmackay9, 2021-09-03:
@yaapu,

It looks like the PR was merged. Do you think we can close this issue?

ojoojoo, 2021-09-03:
Yes sorry, everything is working fine now, the problem was a defective air unit, DJI send it back after repair and now it's working fine.

rmackay9, 2021-09-03:
@ojoojoo, great, thanks!

closed 2021-09-03 by rmackay9