ArduPilot/ardupilot #17816: Rover: PANIC when running DO_SET_MODE mission in SITL

muramura, 2021-06-20:
## Bug report

https://discuss.ardupilot.org/t/do-set-mode-does-really-works/72708
I saw this discus and PANIC when I run the DO_SET_MODE mission in SITL.
GCS continues to display NOLINK.
GDB is outputting PANIC.

**Issue details**

_Please describe the problem_

**Version**
ArduRover V4.1.0-dev (39c96f63)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
Default

**Hardware type**
SITL

**Logs**

rmackay9, 2021-06-20:
I've created an issue here for MP to remove DO_SET_MODE from the list of possible mission commands https://github.com/ArduPilot/MissionPlanner/issues/2652

I'll leave it for others to decide what to do about the PANIC.  My guess is that we want the panic in SITL so that autotest can catch cases where someone tries to use an unsupported command.

muramura, 2021-06-21:
I would like to return the execution result.  I don't think it needs to be panic.  I think it's better for all gcs to return the result with fc than to do it with gcs.

peterbarker, 2021-06-30:
Correct fix here is probably to just add the relevant string!

Even better would be to audit to ensure all are covered - but this is a SITL-only panic to pick up exactly what it did.

PR #18649 (2021-09-13): AP_Mission: complete set of strings and remove support for MAV_CMD_DO_SET_MODE
Fixes #17816

This removes mission support for MAV_CMD_DO_SET_MODE, it is not implemented by any vehicle. Also completes the set of strings for all commands we can save in storage.
closed 2021-09-13 by peterbarker