ArduPilot/ardupilot #17836: Copter: arming via switch enables airmode - but no way to disable

CJ-Davies, 2021-06-22:
## Bug report

**Issue details**

Arming Copter via switch (RCx_OPTION = 41) also activates AirMode in Stabilize flight mode. This is documented behaviour [as per the wiki](https://ardupilot.org/copter/docs/airmode.html).

However there is no way to **disable/prevent** AirMode from activating in this scenario. You cannot use Stabilize without AirMode when arming via switch.

**Version**
4.0.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tested on Y6B, but likely applies to all

**Hardware type**
Orange Cube

**Logs**

IamPete1, 2021-06-22:
IMHO now we have a dedicated switch and param option we should just stop enabling it for switch arming.

andyp1per, 2021-06-22:
Makes sense to me. I guess ideally we would need an option for stabilize as well.

rmackay9, 2021-09-01:
@CJ-Davies, why would you not want airmode?  I would have thought that in general it is helpful the only downside is that it makes it difficult to disarm.

CJ-Davies, 2021-09-01:
> @CJ-Davies, why would you not want airmode? I would have thought that in general it is helpful the only downside is that it makes it difficult to disarm.

A feature that makes disarming difficult shouldn't be forced upon setups that will never actually use that feature. Airmode provides no benefit whatsoever for setups that never perform zero throttle manoeuvres.

PR #18478 (2021-09-20): Copter: do not enter airmode when armed with switch
Fixes #17836

This stops entering airmode with armed with a switch. If you want airmode in ACRO you can set `ACRO_OPTIONS`. Or configure the airmode on/off switch. Or use a interlock switch. 

The current code makes some less keen to use switch arming due to being stuck with air mode. 

We do loose the ability to auto enable in stabilize mode, instead of removing we could put this under `FLIGHT_OPTIONS`. We could add a new arming switch `arm/disarm with airmode`, that might be slightly nicer than adding a `FLIGHT_OPTIONS` bit. 

Quite hard to [... cut]
closed 2021-09-20 by tridge