ArduPilot/ardupilot #17858: Copter/Camera: trigger distance should turn off after exiting Auto (if initiated from Auto)

rmackay9, 2021-06-25:
If the distance based camera triggering is initiated by the DO_SET_CAM_TRIGG_DIST mission command (while in Auto mode) then this triggering should be disabled once the mission is interrupted or completed.

Note that distance based camera triggering can be started in three ways:

1. User can manually set the CAM_TRIGG_DIST parameter
2. DO_SET_CAM_TRIGG_DIST command received from the ground station (although it is not clear if any ground stations support this)
3. Auto mission command DO_SET_CAM_TRIGG_DIST 

The AP_Camera library does not record how it was started and the results all end up directly affecting the CAM_TRIGG_DIST parameter.

This issue was raised during Copter-4.1.0 beta testing. https://discuss.ardupilot.org/t/camera-trigger-does-not-turn-off-after-switching-to-another-flight-mode/72867

ErikSuring, 2021-06-25:
I have used this feature to trigger the camera automatically while free flying in loiter after or while interrupting an auto mission. If this is removed there may not be another way to do this besides manually setting parameters while in flight.

It looks like Andre-K's "CAM_AUTO_ONLY: Distance-trigging in AUTO mode only" parameter may already address this issue. https://ardupilot.org/copter/docs/parameters.html#cam-auto-only-distance-trigging-in-auto-mode-only

Using this parameter also retains distance based triggers when auto mode is reselected. If this parameter is satisfactory I recommend retaining the current behavior.

WickedShell, 2021-06-25:
Yeah as pointed out the auto only option mostly covers this already. (I've also seen some people grab out of a mission with guided to get more coverage, but this is admittedly a bit of a hack and an edge case we could lose safely)

rmackay9, 2021-06-25:
Maybe we should make the parameter *not* be a trigger for starting the distance-based-triggering.  Instead it could just a default distance used whenever the feature is engaged.  The we can setup an auxiliary switch (now available to all vehicles) to start/stop the distance based triggering and auto can keep using the do-set-cam-trigg-dist commands.

MikeCharikov, 2021-06-29:
It also should turn off when entering RTL due to failsafe state.

Gone4Dirt, 2021-11-30:
As has already been mentioned we already have a param to trigger in auto only which was the original request in the forum post.  

An Aux switch function to start/stop is a good idea. So long as triggering in auto ignores the switch position.  It would be annoying to have started some of the mission with the switch accidentally in the wrong position only to have to restart the mission when you realise the mistake...