ArduPilot/ardupilot #17861: Rover: add support for bumpers with simple avoidance

rmackay9, 2021-06-25:
The idea behind this request is that bumpers mounted to a rover will cause the rover to stop and (optionally) backup.

It is possible that the bumpers could be connected as AP_Buttons or a range finder although it might be necessary to have a new driver.

This request comes from this discussion: https://discuss.ardupilot.org/t/object-avoidance-with-bumper/72946

electroklas, 2021-06-26:
Great.. Looking forward to this future addition…
If possible, support for multiple bumpers vould be nice.
In lawnmower applications it would be nice to have individual corner placed bumpers to determine what way to go arround an object. Much like robotic vacuum cleaners.

geofrancis, 2021-09-10:
This has scope to be used on more than just a bumper , it could be connected to any directional sensor that gives a digital output like PIR to give thermal tracking.