ArduPilot/ardupilot #17864: Copter: VTX setting & readout doesn't correspond to transmitted channel

MichelleRos, 2021-06-26:
## Bug report

**Issue details**

When using VTX control via smart audio, the frequency transmitted doesn't correspond to the frequency transmitted: 
This is the readout on the messages:
[image]
And these are the parameters:
VTX_BAND         0.000000
VTX_CHANNEL      0.000000
VTX_ENABLE       1.000000
VTX_FREQ         5865.000000
VTX_MAX_POWER    800.000000
VTX_OPTIONS      0.000000
VTX_POWER        25.000000

But on my goggles, the video turned up on Band A channel 2, which is 5845 MHz

**Version**
4.1-dev (f0d1644a)

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Matek H743-slim

**Logs**
[00000173.zip](https://github.com/ArduPilot/ardupilot/files/6720307/00000173.zip)

andyp1per, 2021-06-26:
Hey @MichelleRos so you set frequency rather than channel and band? What VTX do you have?

MichelleRos, 2021-06-27:
@andyp1per I thought I had set band to 0 (i.e. band A) and channel to 1 and not set frequency, but it seems like that changed somehow (or I forgot to set it). 
It's a TBS Unify Pro Race VTX.

I'm actually a bit unsure about how the parameters work between band & channel vs frequency - If I've set it by frequency and I change the channel, does it auto-change the frequency parameter or what does it do with that parameter?

Thanks. 🙂

andyp1per, 2021-06-27:
It should change the frequency to match the channels you have set and vice versa. If you have set both then it will use the channels. Note that the channel number is indexed from 0 in the settings - maybe we should change this.

MichelleRos, 2021-07-16:
@andyp1per Apologies for the long delay.
I tested different frequencies and didn't manage to reproduce that issue.
I do agree that it may be less confusing to people if they can set the channel parameter to 4 and have it turn up on channel 4 of that band. Of course, changing it may be confusing to people who have gotten used to the offset...

IamPete1, 2021-08-16:
is this still a issue?

MichelleRos, 2021-08-19:
@IamPete1 I was wondering whether we wanted to make the channel number param 1 indexed so that param = 3 would actually mean channel 3, but I guess we may as well close it.