ArduPilot/ardupilot #17871: Compilation error while installing

huseyin-yildiz, 2021-06-27:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

I cant install it today

_Please describe the problem_
it gives an compilation error in  AP_BattMonitor_Logging.cpp file
**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**

[image]

rmackay9, 2021-06-28:
This seems related to this discussion: https://discuss.ardupilot.org/t/sitl-linux-set-up-troubleshooting-error-in-the-ap-battmonitor-logging-cpp/73021
and this PR: https://github.com/ArduPilot/ardupilot/pull/17873

IamPete1, 2021-07-18:
The PR fixing this has now been merged, if the issue remains feel free to re-open.

closed 2021-07-18 by IamPete1