ArduPilot/ardupilot #17878: Flash Write Not Working on STM32H743

yatink-ci, 2021-06-28:
## Bug report

**stm32_flash_write() Function not Writing data when used on STM32H743 (CubeOrange)**

stm32_flash_write() Function in libraries/AP_HAL_ChibiOS/hwdef/common/flash.h, doesn't Work, on STM32H743, I've tested it on CubeOrange, Trying to Flash "Hello World" but it repeatedly fails, and never writes the data in the flash, I've tried Writing to address 0x08020000 and 0x081E0000 but no Luck, I've also tried keeping the flash unlocked during the whole process, but it also doesn't seem to make it work

**Version**
master/4.1.0/4.0.4

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
CubeOrange

**Logs**
N/A

magicrub, 2021-06-29:
This feature works fine. Your custom software is doing something wrong. Please post your branch here so we can see.

yatink-ci, 2021-06-29:
Is it ?
Please have a look at it at - https://github.com/yatink-ci/ardupilot/commit/566b7dd65e5d11da05a6aad5436f062286da99df
Thanks.

yatink-ci, 2021-06-29:
And I've also tested it on CubeBlack, The same Code works on CubeBlack (with Different Flash address Ofcourse) but it doesn't on CubeOrange. I think it has something to do with the H7's Complicated Flash write process.

yatink-ci, 2021-06-30:
Hi, @magicrub , Can you please help in the issue.

yatink-ci, 2021-06-30:
@tridge ? Can you please help with This ? Please ? 🥺
Thanks

IamPete1, 2021-08-08:
Since this is not really a AP issue I'm going to close, feel free to jump on the dev discord if your still stuck.  https://discord.gg/E6Dh2tv8

closed 2021-08-08 by IamPete1