ArduPilot/ardupilot #17879: i2c bus not working with CAN bus enabled

RAZOR15000, 2021-06-28:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_
since this cubes uses here3 GPS, CAN bus is enabled. with that hereflow is also connected and it work perfectly. 
due to that, i2c bus is not working completely for external peripherals. 
such as VL53L1x and beneweke TF luna modules. tf luna module wont work with the serial port as well.
due to this using cube for connect with external rangefinder sensors are completely useless. 
as a bonus issue, i tried using gpio pwm input to get the data from sensors and map it through a arduino. but surprisingly those rangefinder parameters are completely useless, nothing happen when change the parameters. only maximum distance showing is 200cm.  this happening in aux out ports, forum links are below.
https://discuss.cubepilot.org/t/aux-out-pwm-rangefinder-distance-scaling-not-working-in-copter-4-07-cube-orange/6710/2

**Version**
_What version was the issue encountered with_
4.07  and 4.1 beta
**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
QUAD H
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Cube Orange with standard carrier board
**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2021-06-28:
@RAZOR15000,

Txs for the report.  So you're saying that the I2C ports don't work if UAVCAN is enabled on a CubeOrange?  TBH, that seems incredible that we could have such a huge issue and not be aware of it until now.  Especially with 4.0.7... still, we will investigate.

RAZOR15000, 2021-06-28:
yes, its true. i tried with 4.1 beta too its not working.  but in the 4.1 dev i2c is functional for selected devices. such as vl53lx1 and serial tf luna module. but still beneweke tf luna i2c modules don't work in dev version either.

thank you.

rmackay9, 2021-06-28:
Here is a similar report from another user on the cubepilot site: https://discuss.cubepilot.org/t/hereflow-maxbotix-i2c-sonar-issue/1550

tridge, 2021-06-29:
@rmackay9 I just tested an I2C RM3100 compass along with a UAVCAN GPS with no issues.

RAZOR15000, 2021-08-28:
Hello sorry for the delay. Issue was resolved after updating the firmware for the tf luna module, with using the 4.1 dev version. Problem was with old firmware that comes in the tf luna module in the i2c mode. 
Thank you

closed 2021-08-28 by RAZOR15000

aminsteve, 2021-11-05:
This issue is marked as solved, while I am experiencing the same issue with I2C sonar sensors. I am using Cube Orange + Here3 + Maxbotics I2C. Also, my configuration is Copter "Octo X".
- The I2C sensor works independently (checked with RPi and Arduino).
- The I2C sensor works with Ardupilot when the CAN_D1_PROTOCOL and CAN_D2_PROTOCOL are set to 0.
- The I2C sensor stops working when CAN_D1_PROTOCOL and CAN_D2_PROTOCOL are set to 1.

aminsteve, 2021-11-23:
Anybody can help me with this issue? I updated to copter 4.1.1 and this issue is still there.