ArduPilot/ardupilot #17885: Low throtlle stick on QLoiter descent inhibits yaw control

Hwurzburg, 2021-06-28:
stick fully low on QLOITER inhibits yaw...DZ above and it works
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QuadPlane

IamPete1, 2021-06-29:
Do you have a log? Most likely were prioritizing roll/pitch above yaw because motors are saturated low, if that is the case its more or less a hardware limitation.

Hwurzburg, 2021-06-29:
pretty sure its not saturating...here is a log....with default descent and very slow....slower doesn't stop the issue
[00000001.BIN.txt](https://github.com/ArduPilot/ardupilot/files/6733290/00000001.BIN.txt)
renamed to txt to attach

Hwurzburg, 2021-07-03:
this does not happen with Copter only QuadPlane

lthall, 2021-07-09:
[image]

The yaw rate has been commanded.

I will close this for now. Please open it again if we have a log showing the problem with the yaw stick in the neutral position.

lthall, 2021-07-09:
Sorry my bad, this isn't associated with the yaw rate reset issue. This is caused by this line here that sets yaw rate to zero with zero throttle:
https://github.com/ArduPilot/ardupilot/blob/b8e934ccf122dcc5e10cce2953f4bd9d280c9023/ArduPlane/quadplane.cpp#L1600

So the question is, is this the intended behaviour.

Hwurzburg, 2021-07-09:
this was the one that @tridge said we should try qualifying with if the descent rate was => max loiter Z speed, to not zero yaw....remember the landing detection discussion during the dev call?..on my list to try to do

IamPete1, 2021-08-16:
https://github.com/ArduPilot/ardupilot/pull/18021 has been merged, is that a sufficient fix?

Hwurzburg, 2021-08-16:
yep

closed 2021-08-16 by Hwurzburg