ArduPilot/ardupilot #17887: Copter: add ATC_HOVR_ROL_TRM for Tri copter to compensate for lateral thrust from tail rotor

rmackay9, 2021-06-28:
Add ATC_HOVR_ROL_TRM param to allow users to specify a roll angle which compensates for the lateral thrust from the tail rotor.  This would mean that tricopters would be less likely to move laterally when in a level hover.

This is useful for normal pilot controlled flight but also useful if the EKF failsafe triggers and moves the vehicle into Land mode (without position control).

This enhancement request comes from this Copter-4.1 beta testing discussion https://discuss.ardupilot.org/t/sitl-fly-away-in-land-mode-with-tricopter/73001/3

IamPete1, 2021-06-29:
I think we should be able to calculate it automatically. On most tricopters is should not be a noticeable drift, much less torque to fight than helis.

rmackay9, 2021-06-29:
@IamPete1 I suspect you're right because I haven't heard user requests for this new parameter.  Instead this request has come from odd SITL behaviour in which the vehicle moves very rapidly horizontally when in a stable hover.  https://github.com/ArduPilot/ardupilot/issues/17876

IamPete1, 2021-08-16:
I'm going to close, but we should re-open if there is any reports of this issue with real tricopters. It does look like both the heli and tricopter could be fixed with the setting of AHRS_TRIM_X. I guess that looks untidy on the bench.....

closed 2021-08-16 by IamPete1