ArduPilot/ardupilot #17888: Copter: fail to arm if set-home during arming unexpectly fails

rmackay9, 2021-06-29:
As discovered in PR https://github.com/ArduPilot/ardupilot/pull/17875 it is at least theoretically possible for the set-home performed as part of arming to fail unexpectedly which could potentially mean the vehicle flies to an unexpected location far from the takeoff location.

We should fail to arm if the set-home fails unexpectedly.  The solution should involve changing AP_Arming_Copter::arm by replacing the, "// ignore failure", lines below with a "return false;" but it will also require moving this failure higher in the function to ensure that as little vehicle state is changed as possible (e.g. arming LEDs, etc)

```
    } else if (!ahrs.home_is_locked()) {
        // Reset home position if it has already been set before (but not locked)
        if (!copter.set_home_to_current_location(false)) {
            // ignore failure
        }

```

rmackay9, 2021-06-29:
This is surprisingly difficult to resolve actually.  See comments at the bottom of PR https://github.com/ArduPilot/ardupilot/pull/17890

muramura, 2021-07-04:
@rmackay9 san.
Is this bug only in AC4.X?
Or is it a bug in all revisions of Copter?

rmackay9, 2021-07-04:
@muramura,

The bug exists in 4.0 as well (and perhaps older versions too).  It is difficult to trigger in practice though because you need both a a large GPS glitch (>50km) and it must be stable (the GPS position can't be moving around much).  In SITL it is relatively easy to create because of the SIM_GPS_GLITCH_x parameters but in real life it is very difficult.