ArduPilot/ardupilot #17891: Blocking calls in motor output functions slow down fast_loop when using DShot

spiderkeys, 2021-06-29:
## Bug report

Today I discovered that when using DShot, a path of code is triggered in the RC Output chain that can lead to significant blocking within the fast_loop (observed so far only on Sub, but from a quick review, it looks like other platforms might also be impacted).

With 6 motors and running DShot(150-600), I am seeing around 7-8ms getting spent in the motors.output() function of AP_Motors6Dof, as called within Sub's fast_loop(), here:

https://github.com/ArduPilot/ardupilot/blob/8310f4a71881049c251b375de7703ef9cdcf0287/ArduSub/motors.cpp#L21

After instrumenting various levels of the code with custom logging events, I found that applying each motor output consumes roughly 1ms in the delay_microseconds() call in the following downstream method:

https://github.com/ArduPilot/ardupilot/blob/8310f4a71881049c251b375de7703ef9cdcf0287/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L1062-L1071

This causes the 400Hz sub fast_loop() to crawl to a halt, severely impacting performance everywhere. By switching back to basic PWM, all of the issues go away.

Perf log debug output consistently shows such measurements:
```
AP: PERF: 4000/4000 [23259:7335] F=122Hz sd=4869 Ex=600
```

**Version**
819c331acdf4675ac5b6e2b519f1e22f8ed511be

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[X] Submarine
(maybe others as well?)

**Airframe type**
Observed on Vectored and Vectored_6DOF (but likely all are affected)

**Hardware type**
Matek H743-Wing

The photo below shows the time delta across the delay_microseconds() call mentioned earlier, where EV:184 is before, and EV:185 is after. Exact timestamps in microseconds show a delta of ~1ms on average, but I saw instances greater than that.
[image]

andyp1per, 2021-06-29:
On discord you said this only happens at 25Hz loop rate and only on sub?

spiderkeys, 2021-06-29:
I am seeing this behavior with the fast loop rate set to 400Hz (the default in Sub). I did not notice it when I tried to run as close of a configuration as possible in Copter, but my testing may have been flawed, as I am not very familiar with the Copter platform and any behavior differences from Sub.

Here is some additional instrumentation that captures the min_pulse_trigger_us and trigger_widest_pulse values while this is occurring:
[image]

andyp1per, 2021-07-27:
Fixed by https://github.com/ArduPilot/ardupilot/pull/17910

closed 2021-07-27 by andyp1per