ArduPilot/ardupilot #17900: 4.1 Plane Autotune

Hwurzburg, 2021-06-29:
Several new users of 4.1 Plane autotune report that the tune leaves the roll axis wobbling after tune...level4 thru level6 have been reported as being used

logs from users:
Trevellyan: https://www.dropbox.com/s/k6uq4pgb2lkyt6e/log_8_2021-6-10-06-41-14.bin?dl=0
rimb05: https://www.dropbox.com/s/5bdti0dctfc7rat/no%20wind.bin.zip?dl=0, https://www.dropbox.com/s/40o3c5k96xi58ul/wind.bin.zip?dl=0

also on discord, and have requested logs: https://discordapp.com/channels/674039678562861068/797952826847920128/859243883691376661

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

tridge, 2021-06-30:
For the Trevellyan log, the resulting D and P gains are clearly too high and we're getting limit cycle oscillation. The D and P being too high implies that the oscillation detection is failing. 
That is surprising as the 150Hz SMAX isn't causing the Dmod anti-oscillation to trigger. 
Some possible explanations:
 - the servos are much slower than typical servos, or are being run at low voltage
 - the coupling between servo movement and control surface movement is slow, for example with a too flexible foam control surface with the horn at one end of the control surface, or a control rod that isn't stiff enough
I'd like to see some photos or videos of the control surface setup to see if there is something that explains very slow response.
For fixing it, I think a much lower value of RLL_RATE_SMAX would be needed. Maybe 75Hz, but could need as low as 50Hz.

giacomo892, 2021-07-27:
Closed by #18092 ?

Hwurzburg, 2021-07-27:
No confirmation yet

IamPete1, 2021-08-16:
@Hwurzburg How about now? ;)

Hwurzburg, 2021-08-16:
yep think its good to close at this point

closed 2021-08-16 by Hwurzburg