ArduPilot/ardupilot #17901: Passing Aileron output for yaw-authority in QSTABILIZE

gunnir, 2021-06-29:
*Request*: Map aileron servo output to yaw in hover, and switch back to roll for forward flight. 

[image]

Rotating both left/right wings on a fixed axis with a single continuous tilt rotor reduces yaw-authority compared to a vectored yaw setup with independent servos. However, with prop stream over the wings and ailerons, aileron servos can be used to give yaw authority in hover.

With this setup, a single servo can be used to tilt both wings, and aileron servo(s) can be used for yaw in hover, and roll in forward flight. This allows the wings to share a single carbon fibre spar to rotate about, increasing mechanical rigidity and reducing complexity. This is very essential for a 'wing-rotating' tilt rotor, as it is significantly more mechanically complex to rotate them independently on a single axis.

IamPete1, 2021-07-18:
you can do this via scripting, that is probably the best option, more flexible than trying to parametrise it somehow.

IamPete1, 2023-04-24:
This can be done in scripting with https://github.com/ArduPilot/ardupilot/pull/21148