ArduPilot/ardupilot #17921: GCS software (QGroundControl and MissionPlanner) do not update parameter description and value options

tmxkn1, 2021-07-02:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

We customised the description and value options of some existing parameters in a custom firmware by editing the `@Description` and `@Value` in comment area above the parameters in _Parameters.cpp_. However, these modifications are not reflected in GCS software.

**Version**
ArduSub 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[✔] Submarine

**Airframe type**
Sub

**Hardware type**
Pixhawk4

**Logs**
N/A

In specific, we modified The modified [L286-L289](https://github.com/ArduPilot/ardupilot/blob/96882ed0aeeb822dd80575312b782fe47c352b25/ArduSub/Parameters.cpp#L286):

```
    // @Description: Set this parameter according to your vehicle/motor configuration. For MALLARD, use either 7, 8 or 9.
    // @User: Standard
    // @RebootRequired: True
    // @Values: 0:BlueROV1, 1:Vectored, 2:Vectored_6DOF, 3:Vectored_6DOF_90, 4:SimpleROV-3, 5:SimpleROV-4, 6:SimpleROV-5, 7:Custom, 8:Diagonal, 9:Full-Control
```

We expect to see parameter descriptions in MP or QGC to be updated with the "MALLARD" descriptions and value options to include 8 and 9.

Are we doing something completely wrong?

IamPete1, 2021-07-02:
Ground control software pulls descriptions the sever, that data is build from master / release branches. 

https://firmware.ardupilot.org/Tools/MissionPlanner/MetaData/

So you would need to edit the GCS software to point at your custom xml.

It would be quite nice to support packaging xml's in ROMFS so GCS could use FTP to get the file and display the correct info.  Might have to be just the changes from master tho can't use up too much flash.

khancyr, 2021-07-02:
No need to recode anything in the gcs, for mission planner there is https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/generate_mp_paramfile.sh to generate the new xml to load.
I don't remember for QGC but you should be able to manually load the right generated file too

tmxkn1, 2021-07-03:
> No need to recode anything in the gcs, for mission planner there is https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/generate_mp_paramfile.sh to generate the new xml to load.
> I don't remember for QGC but you should be able to manually load the right generated file too

I did come across this as I was searching through the code for other hints of editing parameters, but how do I actually load the generated file into mission planner? I can't find an option in the GUI, except for a button in Setup/Advanced called "Param gen" that seems to be updating parameters from an online source.

tmxkn1, 2021-07-13:
> No need to recode anything in the gcs, for mission planner there is https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/generate_mp_paramfile.sh to generate the new xml to load.
> I don't remember for QGC but you should be able to manually load the right generated file too

Thanks for your reply.

To update parameter meta data in Mission Planner, one needs to generate a new _ParameterMetaData.xml_ using the script you mentioned and replace the one in _%UserProfile%/Documents/Mission Planner_. There is also a _ParameterMetaData.xml_ in MissionPlanner root folder, but replacing that one doesn't seem to work. If the change is small, editing the xml file directly is also a viable option.

This is not a bug of the ardupilot itself, hence closing the issue.

closed 2021-07-13 by tmxkn1