ArduPilot/ardupilot #17922: Forward vision message may flood telemetry link

chobitsfan, 2021-07-03:
## Bug report

**Issue details**
In such setup, companion computer send vision_position_estimate or att_pos_mocap to FC for non-gps flight.
[image]
 Because there is no target_system in these messages, they are considered broadcast messages. FC forward them to other link. Vision messages are usually sent in rather high frequency (~30hz). It may flood some low throughput link (915 telemetry) and make the link nearly unusable. 

A quick fix is to do something like
https://github.com/ArduPilot/ardupilot/blob/c2c013964d419a8965981173042c0f3de37b604d/libraries/GCS_MAVLink/MAVLink_routing.cpp#L108-L112
But it may be useful for GCS to receive forwarded vision messages in lower rate

**Version**
Master

**Platform**
[X] Copter

rmackay9, 2021-07-03:
@chobitsfan, we've added a SERIALx_OPTION to allow not forwarding on messages received to other serial links.  We should probably add this to the wiki if it's not on there already.

chobitsfan, 2021-07-03:
@rmackay9 Thank you very much :)

It would be better if we could find a way to constraint forward rate. But I can not find a easy way to do this

closed 2021-07-03 by chobitsfan