ArduPilot/ardupilot #17926: No LTM Kakute F7

andypnz, 2021-07-04:
## Bug report

LTM (option 25) does not appear to be working on the Kakute F7. Tried multiple serial ports, no data seen on serial analyser.

Changing to eg option 2 - serial data seen. So not a serial port hardware issue.....

4.1.0 Dev on Omnibus F4 - LTM (option 25) working - data seen on analyser.

I cant find an exclusion for the Kakute F7 to not be loading the LTM protocol code?

**Issue details**

Using Latest (4.1.0 Dev) or build from master

**Version**
4.1.0 Dev

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Wing

**Hardware type**
Kakute F7

**Logs**

IamPete1, 2021-07-04:
Its not included on KaKute F7 to save flash, see. https://ardupilot.org/plane/docs/common-limited-firmware.html#firmware-limitations-on-autopilot-hardware

andypnz, 2021-07-04:
In the hwdef file I can see where some features are disabled to save flash - but no mention of LTM….

And I don’t see hal_minimised_features referenced in the LTM code….

Would you be kind enough to let me know where it is disabled?

Thank you!

IamPete1, 2021-07-04:
Include is here:

https://github.com/ArduPilot/ardupilot/blob/c1a5ca6060f16814174e636c5dea470e2f584b1b/libraries/GCS_MAVLink/GCS.h#L975

closed 2021-07-26 by IamPete1