ArduPilot/ardupilot #17932: Plane: Stop plane from diving when RTL with RTL_CLIMB_MIN

RM-FPV, 2021-07-05:
## Bug report

**Issue details**

The plane climbs to RTL_CLIMB_MIN altitude, but then it immediately dives back to the same alt it was before climbing, instead of return to home from highest altitude reached.

Please remove the diving action. It's not safe, as plane can hit an obstacle instead of avoiding it.

**Version**
Latest

**Platform**
[+] Plane

**Airframe type**
Fixed wing

**Hardware type**
Omnibus f4v3

**Logs**

Video.
https://drive.google.com/file/d/1J3aHIHR81eOxvB9mDtXbF5ZTmk3Ja1Wm/view?usp=sharing

RM-FPV, 2021-07-05:
ALT_HOLD_RTL = 170
I would suggest not immediately return to this Alt, but slowly descend like it always does from the the current Alt without using RTL_CLIMB_MIN.
I'll do a log next time.

IamPete1, 2022-03-06:
closed by https://github.com/ArduPilot/ardupilot/pull/16565

closed 2022-03-06 by IamPete1