ArduPilot/ardupilot #17937: Cannot conduct tail rotor control check while disarmed in Copter 4.1

bnsgeyer, 2021-07-05:
**Issue details**

Prior to arming, moving the rudder stick does not cause the tail rotor to respond at all.  Therefore user cannot conduct tail rotor control checks in the disarmed state.  @Gone4Dirt and @IamPete1 discovered this issue and determined that it resulted from PR #17541.  

**Version**
4.1 Beta

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Heli

**Hardware type**
N/A

**Logs**
None

Gone4Dirt, 2021-07-05:
I have had a play in SITL reverting this pr #17541.  Can do TR control checks when disarmed again.  Not sure what the best way to resolve this for heli's is without breaking for multis and quadplanes.  Create a virtual of function AC_AttitudeControl::reset_yaw_target_and_rate() and handle it differently in AC_AttitudeControl_Heli?

tridge, 2021-07-06:
this may affect vectored quadplanes too

rmackay9, 2021-08-14:
Fixed by https://github.com/ArduPilot/ardupilot/pull/17981
this was included in Copter-4.1.0-beta6

closed 2021-08-14 by rmackay9