ArduPilot/ardupilot #17951: Copter: 4.1 is not ignoring pilot yaw during takeoff

rmackay9, 2021-07-07:
Copter-4.1 adds a GUID_OPTIONS parameter that includes a bit to ignore pilot input.  This is exposed to the rest of the vehicle code using the ModeGuided::use_pilot_yaw() method but this is not used in ModeGuided::takeoff_run() probably because this makes use of the more general "auto_takeoff_run()".  We will likely need to extend auto_takeoff_run to accept an "ignore_pilot_yaw" boolean argument (perhaps default it to false).  I suspect that ModeAuto may also require this change (but I haven't checked).

Gone4Dirt, 2021-09-08:
Was this issue created as a feature request?  I am just checking because we are going to be changing the expected behaviour for users who have already used this and there may be some users who reposition yaw/heading on takeoff but have it switched off in the rest of the flight.

rmackay9, 2021-09-08:
@Gone4Dirt,

Guided mode's "ignore pilot input" feature is new for 4.1 but it is slightly incomplete because it doesn't cover the takeoff.

PR #18559 (2021-09-08): Copter: ignore pilot yaw during takeoff with option
Fix #17951

This PR allow "Ignore pilot yaw" bit of GUID_OPTIONS and AUTO_OPTIONS with takeoff.

I tested that "Ignore pilot yaw" works in the following conditions in SITL.
 - takeoff in Guided and Auto mode
 - flying in RTL and SmartRTL mode
closed 2021-09-08 by rmackay9