ArduPilot/ardupilot #17968: Mission Planner not showing getting param Screen After build ardupilot code

ahmetozkaraca, 2021-07-08:
Hello everyone,
I've been flying for about a year using ardupilot and pixhawk. I still haven't been able to solve the error I encountered 2 months ago. When I try to connect with GCS(Mission planner) after building the Ardupilot code, the parameter loading screen does not appear. I encounter the same problem when I try without making any changes in the code I have built. I am building with ./waf in Linux environment. Also, although the parameter loading screen does not appear, I can see that all parameters are loaded on the parameter list screen after waiting for a certain time. There were no problems in flight tests. I've done flights in all modes. My only problem is that when making telemetry connection in Mission Planner, the parameter loading screen does not appear.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk

ahmetozkaraca, 2021-07-08:
I found that the building code have missing parameters. I am adding the parameter list. The first one is my build code's parameters. The second one is ardupilot's latest version parameters. There are 42 missing parameters.

[parametersBuild.xlsx](https://github.com/ArduPilot/ardupilot/files/6783895/parametersBuild.xlsx)
[x6_parametre.xlsx](https://github.com/ArduPilot/ardupilot/files/6783896/x6_parametre.xlsx)

rmackay9, 2021-07-09:
I think the parameter loading screen on MP doesn't appear because the parameters are being loaded in the background, possibly using MAVFTP.  This isn't really an ardupilot flight code issue so I think I will close this.

It might be better to raise this in the mission planner issues list here. https://github.com/ArduPilot/MissionPlanner/issues
 
I'll monitor this thread in case there are any follow-up questions or comments.

closed 2021-07-09 by rmackay9