ArduPilot/ardupilot #17970: [BUG] land_repo_active not being reset in RTL

Tal-Bass, 2021-07-08:
Hello,

When Precision Landing and Land Reposition are both enabled, trying to reposition the drone while it is landing will turn off precision landing. The next time the drone goes into LAND mode, it will attempt to precision land again. However, the next time the drone goes into RTL mode, it will no longer attempt to precision land. Only a full reboot or switching into LAND mode will enable precision landing in RTL mode.

The variable **copter.ap.land_repo_active** keeps track of whether or not the pilot has tried to reposition the drone during landing. This variable gets reset when the drone switches into LAND mode:

[https://github.com/ArduPilot/ardupilot/blob/0986ade905f40c17fb08c07533d4d2de176411b8/ArduCopter/mode_land.cpp#L27](url)

However, this variable does not get reset when switching into RTL:

[https://github.com/ArduPilot/ardupilot/blob/0986ade905f40c17fb08c07533d4d2de176411b8/ArduCopter/mode_rtl.cpp#L13](url)

Is there a reason for this?

If not, can this be changed to reset the variable when switching into RTL also?
This is useful when doing precision landings in RTL mode.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rishabsingh3003, 2021-07-08:
I agree I noticed this too.. But I thought it was by design. Its a easy fix though I think

rmackay9, 2021-07-08:
This definitely sounds like a bug and we should fix it.  thanks for uncovering this.

rishabsingh3003, 2021-07-09:
@Tal-Bass I have assigned this to myself, but feel free to do fix it if you want to raise a PR

Tal-Bass, 2021-07-09:
Awesome thanks for the quick response. I'll go ahead and submit a PR in the next few days.

closed 2021-07-14 by Tal-Bass