ArduPilot/ardupilot #17989: Mamba f405 basic MK3 support

FairyTalesl, 2021-07-10:
I want Ardupilot for Mamba f405 basic Mk3 board. The previous Mk2 version of this card is supported. Barometer is added in Mk3 version but no Ardupilot software. Can you help me with these with a little touch to the Mk2 software?

https://www.diatone.us/collections/basic/products/mamba-basic-f405-mk3-40a-6s-8bit-flight-controller-stack

tenten8401, 2021-07-11:
Working on it right now for one of my quads, in the 20x20 version which uses the same target :)
[image]
If you could help me test once I get an initial hex built that'd be great

FairyTalesl, 2021-07-11:
Ok I'm ready to test I'm looking forward to the hex file

tenten8401, 2021-07-11:
@FairyTalesl this hex should work :)
Based on copter 4.1
[arducopter.zip](https://github.com/ArduPilot/ardupilot/files/6797000/arducopter.zip)
You may or may not have to use mavproxy to change frame_class to 1 before it will work. Try it and see.
[image]
[image]

tenten8401, 2021-07-11:
Actually, use this one instead. Just fixed gyro yaw rotation issue
[arducopter_with_bl.zip](https://github.com/ArduPilot/ardupilot/files/6797035/arducopter_with_bl.zip)

PR #17998 (2021-07-24): Add support for Mamba Basic MK3.5 Boards
This adds support for the Diatone Mamba Basic MK3 line of flight controllers, and possibly more that are using MambaF405US-I2C target.

I've tested this on a quadcopter and it seems to fly fine with all uarts & i2c functional. I'm not sure whether to leave RCIN as NODMA or not because CRSF doesn't work real well without it enabled.
[image]

Closes #17989
closed 2021-07-24 by tridge