ArduPilot/ardupilot #17991: FMT messages missing in log in Ardupilot v4.0.6

vpsoftware-mphibr, 2021-07-10:
Hello Team,

I conduct regular flight tests in AUTO mode in Ardupilot v4.0.6 with QGC. Today I did my regular flight tests. Upon processing the generated logs in Mission Planner I got the FMT messages missing error. I tried uploading the logs on other platforms like Dronee Plotter and UAV log viewer but that didn't work either. I think the log files are corrupted because usually this works for me. What could be the reason for it? 

Following is the log file. I have changed the log file extension to .zip from .bin. You can change it again to try viewing it. Please help.

[log_174_2021-7-10-17-36-18.zip](https://github.com/ArduPilot/ardupilot/files/6795627/log_174_2021-7-10-17-36-18.zip)

amilcarlucas, 2021-07-11:
Copy paste the FMT messages from another 4.0.6 log file, and it should work fine

rmackay9, 2021-07-11:
I think I will close this if that's OK because I think it is more of a support issue than a bug or enhancement.  FYI @peterbarker

closed 2021-07-11 by rmackay9