ArduPilot/ardupilot #17997: No GPS telemetry data in passthrough protocol via FPORT

Seeelefant, 2021-07-11:
## Bug report
Currently I do not get GPS sensor telemetry data in passthrough protocol via FPORT, however all other 14 sensors. In Missionsplanner 21 sats are shown.
[image]
[image]
[image]

**Issue details**

_Please describe the problem_

**Version**
Copter 4.1.0-beta5 30-Jun-2021

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4

**Hardware type**
Pixracer/FrSky R-XSR via FPORT/Taranis XD9+
**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2021-08-08:
@yaapu Another one in your area, I'm not even sure if this is expected to work?

yaapu, 2021-09-07:
@Seeelefant  missed this one, sorry.

in fport GPS would be the ONLY sensor you shoud get, are you running my script?

Seeelefant, 2021-09-13:
Dear @yaapu, yes I am running your script. Finally got this working by using a 3,9k pull-up resistor.

yaapu, 2021-09-13:
great news! glad you got iy working!

closed 2021-09-13 by yaapu