ArduPilot/ardupilot #18016: Log file date/time 'properties' does not respect BRD_RTC_TZ_MIN

andypnz, 2021-07-14:
## Bug report

**Issue details**

When a log file is written to SD card, the date/time properties of the file do not respect BRD_RTC_TZ_MIN

(the date/time data inside the file do reflect BRD_RTC_TZ_MIN)

**Version**
4.1.0 master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Wing

**Hardware type**
KakuteF7

**Logs**
[image]

vierfuffzig, 2021-07-14:
@andypnz afaik as for now the onboard OSD clock panel is the only caller to the get_local_time method.

vierfuffzig, 2021-07-19:
i‘ll close this as log time stamps in UTC is desired behavior. feel free to comment and reopen if required.

closed 2021-07-19 by vierfuffzig