ArduPilot/ardupilot #18109: Documentation conflict

mdhmsmith, 2021-07-25:
Documentation serial ports conflict between:

https://docs.px4.io/master/en/flight_controller/pixhawk4_mini.html

and

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/PH4-mini/README.md

and

https://docs.google.com/spreadsheets/d/1-n0__BYDedQrc_2NHqBenG1DNepAgnHpSGglke-QQwY/edit#gid=912976165

**Version**
4.0.7

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4

**Hardware type**
Pixhawk 4 mini

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2021-07-25:
I'm not really sure why your expecting the px4 serial ports to match the AP ones? Its different firmware, no reason they should be the same.

mdhmsmith, 2021-07-26:
> I'm not really sure why your expecting the px4 serial ports to match the AP ones? Its different firmware, no reason they should be the same.

Thanks for the help!  You just helped me work something out.

closed 2021-07-26 by mdhmsmith