ArduPilot/ardupilot #18119: SkyViper indicator LEDs broken

peterbarker, 2021-07-26:
## Bug report

**Issue details**

Indicator LEDs on SkyViper stay off all the time.

This is probably the recent changes to the `BRD_PWM_COUNT` stuff.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X

**Hardware type**
SkyViper

**Logs**
n/a

tridge, 2021-07-26:
likely needs SERVOn_FUNCTION=-1

peterbarker, 2021-07-26:
```
pbarker@bluebottle:~/rc/flight-data/skyviper-red/logs/2021-04-25/flight1$ grep SERVO.*FUNCTION mav.parm
SERVO10_FUNCTION 0
SERVO11_FUNCTION 0
SERVO12_FUNCTION 0
SERVO13_FUNCTION 0
SERVO14_FUNCTION 0
SERVO15_FUNCTION 0
SERVO16_FUNCTION 0
SERVO1_FUNCTION  36
SERVO2_FUNCTION  33
SERVO3_FUNCTION  34
SERVO4_FUNCTION  35
SERVO5_FUNCTION  0
SERVO6_FUNCTION  0
SERVO7_FUNCTION  0
SERVO8_FUNCTION  0
SERVO9_FUNCTION  0
pbarker@bluebottle:~/rc/flight-data/skyviper-red/logs/2021-04-25/flight1$ 
```

peterbarker, 2021-09-03:
This is fixed in master.

closed 2021-09-03 by peterbarker