ArduPilot/ardupilot #18138: fractional turns on loiter

montathar111, 2021-07-27:
I would like to be able to define degrees of the loiter. For example when i set param1 (turns) 0.5f ( 180°) should fly only half of the circle. If param1 should be integer, can we use param2 to define incomplete circles (For example 180 for half of the circle) ?

**Platform**

[  ] Copter

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2021-07-27:
At the moment the way we store the mission commands means that the first field ("Turns") is an integer and so we can't store fractions.  We do have a plan to improve the way we store missions and that could fix this issue but I can't promise when we will do that because it is quite a significant change.

montathar111, 2021-07-28:
So what do I do if I want to make half a turn or quarter of a turn. I really need to do this in my mission.

xfacta, 2021-07-29:
Until there's a specific solution or fix, can you set the craft to face next waypoint (WP_YAW_BEHAVIOR), and set a waypoint at the correct angle from the last waypoint but less than a meter away? So hopefully you can have the craft rotate but with minimal lateral movement.
I'll try it in SITL

xfacta, 2021-07-29:
OK error on my part, I see what you are asking for now - something like the Circle mode but with a partial circle. Sorry I dont have any suggestions for that, but I did try out what I _thought_ you were asking at first - yawing a number of degrees.

_My suggested method didnt work, the second waypoint has to be outside the WP_NAVRADIUS for the aircraft to change yaw, or it just moves over to the new waypoint without turning.
I did have some success with CONDITION_YAW then a DELAY to allow the copter plenty of time to yaw then wait._

xfacta, 2021-07-29:
Auto WP -> Create Spline Circle did a pretty good job, I just deleted some of the waypoints to leave a half-circle, and delete the ROI that gets created if you want the vehicle to face along the track instead of the ROI.
Probably not highly accurate though, depends what you need.

closed 2024-01-23 by magicrub