ArduPilot/ardupilot #18143: Cygwin on Win7-64bit doesn't work as described

olliw42, 2021-07-28:
I'm trying to compile ArduPilot on a Win7 64 bit computer
with 4.0 it did work great for me following the instructions on https://ardupilot.org/dev/docs/building-setup-windows-cygwin.html#building-setup-windows-cygwin at the time.
with 4.1 it seems some things have changes and are still continuously changing, anyhow I can't compile anymore as of today with latest branches

I (re)installed cygwin exactly as described in the docs, i.e., copied the setup line to a console and ran. I was surprised to see it opens the same dialog it normally would do and the packages it shows I'm not sure are those one wants. Anyway. I then (re)installed also the arm compiler. 

./waf distclean gives "Segmentation fault (Speicherabzug geschrieben)"
./waf configure --board  matekh743 just doesn't do anything
./waf copter also gives "Segmentation fault (Speicherabzug geschrieben)"

it would be fantastic if things (scripts&docs) could be updated (soon?) to work again.

many thx

[image]

magicrub, 2021-07-28:
We should update the docs. With WSL available, there's no good reason to use cygwin any more.

olliw42, 2021-07-28:
unless one is still on win7 ... unfortunately :)

magicrub, 2021-07-28:
There's a solution to that. We also don't compile on windows 95.

olliw42, 2021-07-28:
well, there is not point to argue here ... the line of reasoning is generally silly as entirely subjective 

fact appears to be that the docu seems not to work ... if it is ardupilot's intention to remove support so it at least should do this, indeed

khancyr, 2021-07-28:
Did you try with the install script we provide? It should up to date and tested as it mostly match our cygwin CI.

But yes we intend to stop supporting cygwin as it is broken every months. Could you consider using vagrant or docker ? I am not sure what works on Win7.... 

Unfortunately, we are already short in dev time so we cannot support everything forever.
I will update the doc as some explanations make no sense, thanks for raising this

IamPete1, 2021-07-28:
I can build master with Cygwin64, so it is possible. I followed the install instructions at when I set it up, that was probably last year....

olliw42, 2021-07-29:
> Did you try with the install script we provide? It should up to date and tested as it mostly match our cygwin CI.

no, but I looked into it and couldn't see anything which would make me think it would do better/different

> But yes we intend to stop supporting cygwin as it is broken every months. Could you consider using vagrant or docker ? I am not sure what works on Win7....

well, I have to go with what is possible ... at the time being cygwin32 works for me
it's just that cygwin64 is said by the docs to work ... so I thought I would probably be smart to do as it says LOL

> But yes we intend to stop supporting cygwin

I'm not trying to dispute this. It however would be then indeed appropriate to adapt the docs to reflect this

> I can build master with Cygwin64, so it is possible. I followed the install instructions at when I set it up, that was probably last year....

interesting ... I won't claim I'm the smartest then it comes to OS things, but I would think that the steps are simple enough to reproduce correctly ... so, the only conclusion I can draw is that there is some crucial difference in our two systems or what we did which is not mentioned or realized

anyway, if you plan to stop cygwin, pl change the docs

I guess all is said on the topic, so pl close it if you feel alike

IamPete1, 2021-09-12:
@olliw42 did you get any further this this? There have been a couple of Cygwin changes recently https://github.com/ArduPilot/ardupilot/pull/18569. It might be worth trying  `python3 ./waf` if you don't have python three you can get it from the Cygwin installer package manager thingy.

olliw42, 2021-09-13:
no. I understood the above to indicate that cygwin won't be at the forefront of maintance any further. I use cygwin 32bit and it works.

rmackay9, 2021-09-13:
I also still use cygwin because I compile within eclipse so I don't agree with TomP's comment above, "there's no good reason to use cygwin any more".

magicrub, 2021-09-13:
@rmackay9 there's also no good reason to use eclipse either :) VSCode!!

IamPete1, 2021-09-13:
@olliw42 if your happy on Cygwin 32 and no one else has re-produced this I think we might as well close.

We do still need Cygiwn to build the mission planner SITL executables so it should still work for the foreseeable future. But I doubt the instructions or install script will get much love. In general were trying to recommend WSL for windows 10 devs.

closed 2021-09-13 by IamPete1

olliw42, 2021-09-14:
> I guess all is said on the topic, so pl close it if you feel alike

:)

rmackay9, 2021-09-14:
@magicrub,

If we were really going to promote VSCode over eclipse I suspect more work needs to be put into the wiki page. https://ardupilot.org/dev/docs/editing-the-code-with-vscode.html