ArduPilot/ardupilot #18157: Copter: waypoint movement

ForesterDev, 2021-07-29:
## Feature request

**Is your feature request related to a problem? Please describe.**
Improvement of waypoint movement

**Describe the solution you'd like**
Currently waypoint movement for copter based on direct vector from current position and target. So if copter is near the the X/Y but is far from altitude - it will not run horizontal and only have vertical motion. I need a mode, when copter will fly horizontally until it reaches the target, and then change altitude vertically if it has not reached an altitude earlier

**Describe alternatives you've considered**
[image]

I tried to show graphically. On first image - the current behavior, and on second - what i need

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2021-07-29:
Can you not just add a intermediate waypoint?

rmackay9, 2021-07-29:
I've updated the title to try and make it a bit clearer.

I wonder if you could clarify the situation this would be useful?  In general I don't think this is a good method because the operator won't know in advance the path that the vehicle will take.  Depending upon the environment (wind, battery voltage, payload weight) the path could change... I also don't think that the vehicle will reach its destination any faster.

.. still, thanks for the request.  It is interesting to think about.

lthall, 2021-07-29:
Are you able to use guided mode to achieve this?

lthall, 2021-07-30:
This is the behaviour you will see using guided pos, vel, accel input and keeping the vel and accel set to zero.

IamPete1, 2022-09-10:
No response on suggested alternatives, so I'm going to close.

closed 2022-09-10 by IamPete1