ArduPilot/ardupilot #18165: Add an option in the parameters for Surface tracking.

zhouzhiwen2000, 2021-07-30:
## Feature request

**Is your feature request related to a problem? Please describe.**
I am using the drone indoors and I am using the rangefinder as the height source. I don't want the surfacetracking feature enabled to mess with the height target; however, in order to disable it, I have to assign a switch just for this. Is it possible to add an option for surfacetracking in the parameters.

**Describe the solution you'd like**
An option for surfacetracking in the parameters.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

shiv-tyagi, 2021-09-19:
Hey! I tried to work on this issue. I added a parameter for the same and used it to initialise the surface property of surface tracking class. Please review if I did it correctly.

zhouzhiwen2000, 2021-09-24:
Maybe start a pull request? The code looks fine to me, but I myself is new to ardupilot code.

IamPete1, 2021-11-13:
closed by https://github.com/ArduPilot/ardupilot/pull/18716

closed 2021-11-13 by IamPete1