ArduPilot/ardupilot #18169: EKF3 variance when disarmed for long periods of time

NickNunno, 2021-07-31:
## Bug report

**Issue details**

When rover is left disarmed for long periods of time EKF3 seems to get upset and fails to arm due to "PreArm Check: EKF3 Roll/Pitch inconsistent by X degree" 

- If you disable arming checks, the EKF instantly corrects itself when you arm. 
- When you disarm, condition returns. 

[VIDEO](https://youtu.be/lU3e-R1rVcc) of arming reset

[VIDEO](https://youtu.be/WNtmjp7wpxo) of MP EKF status

**Version**
Rover Master - July 6th - 6ec221625b

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
Ackermann

**Hardware type**
Cube Orange, Dual F9 GNSS + Heading 

**Logs**
Unable to retrieve logs since the vehicle needs to sit for many hours.

rmackay9, 2021-07-31:
@NickNunno,

Thanks for this.  To me, this seems similar to [this Copter discussion](https://discuss.ardupilot.org/t/ekf3-position-still-going-mad-in-beta5-drone-crashed/73859/3) and I think since then @priseborough has made significant improvements in the EKF3 stability and this is all after July 6th.

I'm sure you're very busy but I wonder if you could try and recreate the issue using Rover-4.1.0-beta6 and set LOG_REPLAY = 1 so we can run the log through replay to recreate the issue?

NickNunno, 2021-07-31:
Thank you. We were waiting to update since there are so many open yaw related bugs on current master. 

Is this the fix you're referring to? 

https://github.com/ArduPilot/ardupilot/pull/18008

We could cherry pick and test?

rmackay9, 2021-07-31:
@NickNunno,

Yes, that's the fix.  Using -beta6 or master is best because it removes any questions about what code is being used but we will take what we can get!

tridge, 2021-08-02:
@nick2204 if the issue still happens with the new beta then we'll need a replay log to track it down:
 - LOG_DISARMED=1
 - LOG_REPLAY=1

nick2204, 2021-08-02:
> @nick2204 if the issue still happens with the new beta then we'll need a replay log to track it down:
> 
> * LOG_DISARMED=1
> * LOG_REPLAY=1

Think you meant @NickNunno 😉

rmackay9, 2021-08-14:
@NickNunno,

I left my AION robotics rover sitting still running Rover-4.1.0-beta7 running for 5 hours and nothing bad happened.  The EKF was perfectly happy so I think this issue was resolved as part of the EKF3 improvements that went into beta6.

I'm going to close this but we can re-open if you see it again with a later version of Rover-4.1.x.

closed 2021-08-14 by rmackay9