ArduPilot/ardupilot #18196: 60ms delay reading IMU data

tridge, 2021-08-02:
We have 2 logs showing a delay reading IMU data on a MatekH743. The original symptoms were a PM.MaxT over 60ms occurring twice in one flight.
The issue was reproduced on a 2nd board, with the monitor code modified for a lower threshold for the MON message. That gave this result:
```
2021-08-01 00:09:04.70: MON {TimeUS : 432335999, LDelay : 66, Task : -3, IErr : 0, IErrCnt : 0, IErrLn : 0, MavMsg : 0, MavCmd : 0, SemLine : 0, SPI : 4777731}
```
that shows a 66ms delay while waiting for IMU data (task -3)

chobitsfan, 2021-08-09:
Maybe it is related to https://github.com/ArduPilot/ardupilot/issues/16478#issuecomment-774693804? Both MatekH743 and mRo Pixracer Pro use H743.

rmackay9, 2021-08-19:
I think this is now a duplicate because we have issue https://github.com/ArduPilot/ardupilot/issues/18383

closed 2021-08-19 by rmackay9