ArduPilot/ardupilot #18201: Copter: single and coaxcopter fins not centered if using DShot

rmackay9, 2021-08-03:
Similar to the issue resolved for Tricopter in PR https://github.com/ArduPilot/ardupilot/pull/18181, Single Copters and CoaxCopter fins will not be centered if the vehicle is setup to use BLHeli/DShot for the motors.  The issue stems from the get_motor_mask() method returning bits corresponding to the fin outputs.  This bitmask is then used by the BLHeli/DShode code which then overwrites the SERVOx_MIN/MAX/TRIM values.

Hopefully the solution is as simple as removing the fins from the motor mask.  The motor mask's original use was to protect against multiple features attempting to use the same servo output.  We included the fins in the motor mask so that these outputs wouldn't be accidentally used by another feature.  The addition of SERVOx_FUNCTION means that motor_mask is not really required for this protection anymore.

The motor_mask is now used for these things:

- Copter::init_rc_out() uses it to say which outputs may move while the safety switch is engaged.  This means that the fins will likely move while the vehicle is disarmed
- AP_AdvancedFailsafe_Copter::setup_IO_failsafe() uses it so that the motors are at min during an I/O board failsafe.  This change seems ok because we probably don't want the fins going to min during and I/O board failsafe.
- Copter::esc_calibration_setup() enables the output to ESCs for calibration.  This change also seems OK because we don't want to try and calibrate the fins.
- AP_Arming_Copter::motor_checks() uses the mask if ToshibaCAN ESCs are used to ensure enough ESCs are sending feedback.  This change is fine because we wouldn't use ToshibaCAN ESCs for controlling fins

prathamesh7255, 2023-01-21:
@rmackay9  is this issue open or it's closed as I see merged PRs.

IamPete1, 2023-01-21:
@prathamesh7255 Not fixed, we need a patch similar to https://github.com/ArduPilot/ardupilot/pull/19202 for single-copter. 

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsSingle.cpp

UbhayAnand1, 2023-01-27:
I took a shot, by refering to the mentioned file, and created a pull request. @IamPete1 If my PR is not entirely right, but am going in the right direction, can I be assigned this issue please?

UbhayAnand1, 2023-01-27:
My PR failed a lot of tests. I forked the code and when open the files locally, a lot of header files are missing. Can someone please help me out here?

IamPete1, 2023-01-29:
Fix is now complete thanks to, https://github.com/ArduPilot/ardupilot/pull/22762

closed 2023-01-29 by IamPete1