ArduPilot/ardupilot #18240: Sub: 4.2.0 issues list

Williangalvani, 2021-08-06:
This a list of blocking issues/features for Sub 4.2

#### EKF related:
 - [ ] Yaw drift issue (#18229)
 - [ ] DVL integration (ExtNav):
      - [ ] Finish [companion integration](https://github.com/bluerobotics/companion/pull/355)
      - [ ] Get position hold to work smoothly
      - [ ] Deal gracefully with loss of bottom-lock.
      - [ ] (tentative) DVL+GPS operation

#### Control:
 - [ ] Fix depth-hold behavior with non-neutral buoyancies/payload. #15247
 - [ ] Allow holding depth in arbitrary attitudes
#### Misc:
 - [ ] #12780
 - [ ] Send -1 battery_remaining when battery capacity is zero https://github.com/ArduPilot/ardupilot/pull/13712
#### Navigator support:
  - [ ] Set BARO_EXT_BUS default to 6
  - [ ] Add default Power Module to [AP_BattMonitor](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BattMonitor/AP_BattMonitor_Analog.h#L48)

ES-Alexander, 2022-08-25:
# To add
**Navigator support**: 
- relays

**Servo controls**: 
- allow mapping `servo_n_*` joystick functions to arbitrary servo outputs, like with relay pin assignments.

ES-Alexander, 2026-04-06:
@Williangalvani noting that #31109 has been merged into `master` now, but will need to be back-ported to `ArduPilot-4.7`.