ArduPilot/ardupilot #18252: Existing M8030 based GPSes do not work in 4.1

Hwurzburg, 2021-08-08:
in Stable and previous these GPS work well:
https://usa.banggood.com/HGLRC-8M-M8N-GPS-Module-for-APM-Pixhawk-CC3D-Naze32-F3-Flight-Control-for-RC-Drone-FPV-Racing-p-1078657.html
https://www.amazon.com/gp/product/B08636HLFG/

they are limited to 115Kb

in 4.1 they are not recognized...same board setup, just firmware upgrade....apparently the autobauding is not working properly

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Hwurzburg, 2021-08-17:
fixed...closing

closed 2021-08-17 by Hwurzburg

rmackay9, 2021-08-25:
@Hwurzburg do you think we need to somehow force this option bit on for some boards or some GPSs?  I'm just wondering how users will find this fix.

Hwurzburg, 2021-08-25:
I don't know that there is a way to tie this to any board. It's just a inexpensive  gps used in the hobbyist market. I will put a note on the GPS landing page.