ArduPilot/ardupilot #18263: Standard communication protocol between smart battery and flight controller

ZLL6, 2021-08-09:
This is a communication protocol widely used in smart batteries and flight controllers. A large number of smart battery developers and more flight controller applications are using this protocol.
      First of all, this agreement is based on the UAVCAN v0 frame format, which to a certain extent recognizes and supports UAVCAN. Considering the practicability of data and the processing pressure of the flight controller on the node, we have performed necessary and non-essential screening of the data sent by the smart battery. Our protocol is for the flight controller to send a periodic frame. The necessary information includes The total voltage of the battery and the partial voltage of each battery (compatible with any number of battery cells), charge and discharge current, temperature (compatible with any number of temperature points), and the SOC, SOH, design capacity and remaining capacity of the power capacity surface, the last is an error information. For more details, please see the link below
[https://discuss.cubepilot.org/t/standard-uav-intelligent-battery-uavcan-communication-protocol/6986](url)

hendjoshsr71, 2021-08-16:
A couple comments on your document. You should look at the mavlink [SMART_BATT_INFO ](https://mavlink.io/en/messages/common.html#SMART_BATTERY_INFO)message. The purpose of this message is to break out information that only needs to be gathered on boot or request. You are wasting bytes on this info. by combining this info with current voltages, temperature, etc. 

Screenshot below since it is in the link above.
[image]

There are a few fields above that are using limited ranges given the underlying type. See the SMART_BATT_INFO message for inspiration for sizes and additional fields of interest for batteries. 

Capacity should be 32bit as 16 bit with mAh is a 65A-h battery. You can't change a message once it is set so it must be useable into the future and for a variety of vehicle sizes.

ZLL6, 2021-08-23:
> A couple comments on your document. You should look at the mavlink [SMART_BATT_INFO ](https://mavlink.io/en/messages/common.html#SMART_BATTERY_INFO)message. The purpose of this message is to break [...]

Thank you for pointing out, it has been corrected
[https://discuss.cubepilot.org/t/a-smart-battery-communication-protocol/7090](url)