ArduPilot/ardupilot #18268: Preloaded Terrain no longer valid

stephendade, 2021-08-10:
This occurs on latest master.

With the recent accuracy improvements in lat/lon estimation (I think?), the terrain file format is no longer aligned. There's at least 10m delta (used to be 2cm, measured in Canberra) between the terrain grid generated by ArduPilot and the files generated by https://terrain.ardupilot.org/

To replicate:
 - Download a terrain file set from https://terrain.ardupilot.org/ and put in the "terrain" folder for SITL
 - Run SITL with ``TERRAIN_DEBUG 1``
 - Note the message 
   ```
   read empty block at -353747690 1491548424 ret=2048 (-353747690 1491554539 100 0xffffffffffffff) 0x1888:0x1888
   ```

This shows the delta between the grid location in ArduPilot (-353747690 1491548424) and the pre-generated terrain file (-353747690 1491554539). The threshold for this is currently 2cm, above which ArduPilot rejects the pre-generated terrain file and starts requesting a new one from the GCS.

Raising the threshold to 10000 (10m) in ```TERRAIN_LATLON_EQUAL``` (at least in Canberra) allows ArduPilot to continue using the pre-generated terrain without issue.

In terms of resolving this, I see 3 possible solutions:
 - Raise the threshold, noting that there'd be a small delta in (10's of m?) in position used for calculating terrain height
 - Upversion the terrain file format, re-generate the terrain files at https://terrain.ardupilot.org/. Then there'd be the issue of how to tell users which file format version to use.
 - Add some sort of offset in AP_Terrain to account for the grid differences

tridge, 2021-08-10:
nice sleuthing!
let's raise the the threshold for now, then fix the terrain server to match the more precise maths after 4.1 is stable

0110100100101010, 2021-08-10:
6ee9e445a90b7f9c1ba73349abc2326fa6c09091 improved the offset and distance calcs used by calculate_grid_info (and others), while create_terrain.py still calculates the easting at the initial latitude.

Problem lines are:
https://github.com/ArduPilot/ardupilot/blob/fadadfb50e831eadef1a709a149d1ad7b683dbde/libraries/AP_Terrain/tools/create_terrain.py#L55
https://github.com/ArduPilot/ardupilot/blob/fadadfb50e831eadef1a709a149d1ad7b683dbde/libraries/AP_Terrain/tools/create_terrain.py#L60

The create_terrain tool and website should be modified to use the less-wrong method now used in AP_Common/Location.cpp and the block header version bumped. Backwards compatibility with old terrain files could then be implemented in AP_Terrain, and attempts to use newly generated terrain files in old ArduPilot will fail due to the version field.

Raising the threshold arguably makes things worse, rather than failing somewhat loudly and refusing to use the old terrain file it'll now silently preform the lookup with a significant offset and return incorrect heights. 10's of meters XY offset is not minor and depending on the terrain could well be catastrophic.

To be clear info.idx_y and info.frac_y as returned by calculate_grid_info are used directly to index into the heights and scale the interpolation respectively. They no longer reference the same grid as the pre-generated files, and when used with them **AP_Terrain::height_amsl() will return incorrect heights**.

stephendade, 2021-09-02:
This has now been fixed via:
-Addition of the TERRAIN_MARGIN parameter
-Updated terrain tiles now available at https://terrain.ardupilot.org/

closed 2021-09-02 by stephendade