ArduPilot/ardupilot #18283: BattMonitor MPPT weird assignement

khancyr, 2021-08-11:
## Bug report

**Issue details**

Looking at https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BattMonitor/AP_BattMonitor_MPPT_PacketDigital.cpp#L229

we set sequence to 0, then do some assinement to txFrame.data[0] with a if condition.
Next we override txFrame.data[0] with sequence that is always 0. 
That make no sense.

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

magicrub, 2021-08-11:
Yup, that's wrong. Good catch! Luckily, it has no impact on the data, it's a tool to see if we're dropping packets just like MAVLink does.

magicrub, 2021-08-11:
@khancyr  fixed in PR https://github.com/ArduPilot/ardupilot/pull/18296

closed 2021-08-12 by tridge