ArduPilot/ardupilot #18285: Ringbuffer peekbytes function is broken

khancyr, 2021-08-11:
## Bug report

**Issue details**

Looking at https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/utility/RingBuffer.cpp#L171

We can only have access to vec[0]
the issue comes from peekiovec() https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/utility/RingBuffer.cpp#L132

The return value is controled by the len/n comparison.
Let's have simple examples : 
len = 2
n = 1

len > n == true
so set len = n = 1
n > len == false

len <= n == true
then return 1

example two the opposite
n = 2
len = 1

len > n == false
n > len = true
so set n = len = 1
len <=n == true
then also return 1

So the return 2 is unreachable

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2021-08-11:
i wonder if we could rename the file from RingBuffer.h to ByteBuffer.h (or the other way around).  I found it slightly harder than it should have been to find where this class was being used.

peterbarker, 2021-08-12:
> i wonder if we could rename the file from RingBuffer.h to ByteBuffer.h (or the other way around). I found it slightly harder than it should have been to find where this class was being used.

This has bugged me for a while....

WickedShell, 2021-08-12:
> > i wonder if we could rename the file from RingBuffer.h to ByteBuffer.h (or the other way around). I found it slightly harder than it should have been to find where this class was being used.
> 
> This has bugged me for a while....

Also while dreaming: Why is it in in HAL utility? It maybe made sense originally, but we use it in a lot of places that aren't HAL related. And we really don't expect it to be replace by the HAL. It's another one of those things that makes it a bit harder when trying to find it, as you just don't expect to go there.... But this is all separate from the issue at hand :)

khancyr, 2021-08-13:
If somebody can help me fixing the issue, would rename and move at the same time.

PR #18798 (2021-09-30): AP_HAL: added peekiovec test
fixes #18285
commit aa05629e (2021-09-30): AP_HAL: added peekiovec test

fixes #18285
closed 2021-09-30 by peterbarker