ArduPilot/ardupilot #18297: Copter 4.1.0: Dataflash logs downloaded with Mission Planner through USB are corrupted starting from 2nd

mbuzdalov, 2021-08-11:
## Bug report

**Issue details**

When I do several flights on a Pixracer quad, and then download all the dataflash logs by connecting it to Mission Planner through USB, the first of these logs is OK, but all subsequent ones are reported to be broken, and usually indeed are (very strange discharge curves, bad timing etc). These same logs are however OK on the SD card attached to the Pixracer.

Occasionally, when re-downloading certain failed logs, their new copies appear to be OK.

Through the user interface of Mission Planner, the broken logs always manifest themselves by having "Retransmitting XXX" in the process of downloading such a log. Lots of stuff happens in the console, an example is attached.

(This might be actually a Mission Planner issue, however, at least it started to happen when I switched to 4.1.0.)

**Version**
ArduPilot: Copter 4.1.0.beta5, 4.1.0.beta6, and some earlier beta and dev releases too.
Mission Planner: git latest on Gentoo Linux.

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad H

**Hardware type**
Pixracer

**Logs**
Two archives with the logs (pairs of the correct log and of the broken one) are attached, one for beta5, another for beta6. For the sake of making archives smaller I did not include successful log downloads (always the first one). The console dump, also attached, corresponds to downloading the logs from beta6.

[beta5.tar.gz](https://github.com/ArduPilot/ardupilot/files/6971710/beta5.tar.gz)
[beta6.tar.gz](https://github.com/ArduPilot/ardupilot/files/6971719/beta6.tar.gz)
[apmlog_20210811.txt](https://github.com/ArduPilot/ardupilot/files/6971726/apmlog_20210811.txt)

rmackay9, 2021-08-11:
fyi @meee1

mbuzdalov, 2021-08-12:
Sorry, I might have missed the software names. I use APM Planner (compiled from git master).