ArduPilot/ardupilot #18323: AP_Flashstorage: erase_all issue

khancyr, 2021-08-13:
## Bug report

**Issue details**

On AP_FlashStorage::erase_all we try to do 
erase_sector(0, current_sector!=0) but current_sector is always 0. Therefore the check doesn't make sens, and could a be an issue.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_FlashStorage/AP_FlashStorage.cpp#L351

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

tridge, 2021-09-15:
in re_initialise() why can't current_sector be non-zero?

khancyr, 2021-09-15:
@tridge because https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_FlashStorage/AP_FlashStorage.cpp#L348, we always set it to 0 in all case...

khancyr, 2021-09-29:
As review on devcall, the logic is fine, it is just not nicely written.