ArduPilot/ardupilot #18324: COMPASS_DEV_IDx does not match COMPASS_PRIOx_ID

KarolNi, 2021-08-13:
## Bug report

We have an issue with COMPASS_DEV_IDx. When read out they do not match COMPASS_PRIOx_ID. We are afraid that it may affect calibration parameters allocation or compass calibration process. We want to use HERE via UAVCAN as first mag and another HERE via serial+i2c as second mag. We set one on UAVCAN to static ID of 64 (because of dynamic ID was not consistent across our fleet). Because of this we set:
```
COMPASS_PRIO1_ID,81923
COMPASS_PRIO2_ID,592905
```
and expect
```
COMPASS_DEV_ID,81923
COMPASS_DEV_ID2,592905
```
but gets
```
COMPASS_DEV_ID,592905
COMPASS_DEV_ID2,81923
```
It was reproduced on two aircrafts. Mission planner compass tool shows that everything is correct (non missing, correct order) and status is as expected - when touching magnet to UAVCAN one ```mx``` goes haywire, when touching magnet to I2C one ```mx2``` goes haywire.
Other parameters:
```
COMPASS_DEV_ID3,0
COMPASS_DEV_ID4,0
COMPASS_DEV_ID5,0
COMPASS_DEV_ID6,0
COMPASS_DEV_ID7,0
COMPASS_DEV_ID8,0
COMPASS_ENABLE,1
COMPASS_EXTERN2,1
COMPASS_EXTERN3,0
COMPASS_EXTERNAL,1
COMPASS_OPTIONS,0
COMPASS_TYPEMASK,512959
COMPASS_USE,1
COMPASS_USE2,1
COMPASS_USE3,0
```

Similar issues: #16309, #7970
Last commit that seems relevant https://github.com/ArduPilot/ardupilot/commit/9553bc14d01e61f39c60bc1aa611326bdefee7e0#diff-81e39479bd0b9020729cddd75aca3dbb43388c1b9598e44e0f3ef398917df0ec

**Version**
Arduplane-4.0.6(036ad450)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
CubeBlack

IamPete1, 2021-09-12:
@KarolNi can you test on 4.1 and see if this has already been fixed.

KarolNi, 2021-09-13:
I have just tested it on arduplane V4.1.0beta7 (4bf32a5f) and it looks the same on the parameters side, so no further testing. I have done single test (one avionics set on desktop, not on the flying fleet; via MissionPlanner).
|||
|--- |--- |
|COMPASS_DEV_ID|592905|
|COMPASS_DEV_ID2|81923|
|COMPASS_DEV_ID3|0|
|COMPASS_DEV_ID4|0|
|COMPASS_DEV_ID5|0|
|COMPASS_DEV_ID6|0|
|COMPASS_DEV_ID7|0|
|COMPASS_DEV_ID8|0|
|COMPASS_PRIO1_ID|81923|
|COMPASS_PRIO2_ID|592905|
|COMPASS_PRIO3_ID|0|

Since writing ticket we have done couple flights (and even one compass calibration) and there were no crashes, but for me it is still "it works, but nobody knows why" situation.
One aircraft in our fleet is even weirder with:
After setting manual ID on CAN HERE:
```
COMPASS_DEV_ID,0
COMPASS_DEV_ID2,592905
COMPASS_DEV_ID3,81923
COMPASS_DEV_ID4,0
COMPASS_DEV_ID5,0
COMPASS_DEV_ID6,0
COMPASS_DEV_ID7,0
COMPASS_DEV_ID8,0
```
Then after compass calibration (via MissionPlanner):
|||
|--- |--- |
|COMPASS_DEV_ID|0|
|COMPASS_DEV_ID2|592905|
|COMPASS_DEV_ID3|0|
|COMPASS_DEV_ID4|0|
|COMPASS_DEV_ID5|0|
|COMPASS_DEV_ID6|0|
|COMPASS_DEV_ID7|0|
|COMPASS_DEV_ID8|0|

But when checked last flight log (using plot.ardupilot.org) everything is as intended (but MissionPlanner full parameter list still shows as above):
```
COMPASS_DEV_ID,81923
COMPASS_DEV_ID2,592905
COMPASS_DEV_ID3,0
COMPASS_DEV_ID4,0
COMPASS_DEV_ID5,0
COMPASS_DEV_ID6,0
COMPASS_DEV_ID7,0
COMPASS_DEV_ID8,0
```
All systems above were set all the time as follows:
```
COMPASS_PRIO1_ID,81923
COMPASS_PRIO2_ID,592905
COMPASS_PRIO3_ID,0
```

IamPete1, 2021-09-13:
@bugobliterator I wonder if this is the expected behavior?

KarolNi, 2021-09-30:
I have further update about one aircraft (the "even weirder" one). We track parameters of every aircraft in our private repo (committing usually from airfield, or with big configuration change). Today we where doing some ground testing of this one and this diff showed up:
```
 arduplane.param | 46 +++++++++++++++++++++++-----------------------
 1 file changed, 23 insertions(+), 23 deletions(-)

diff --git a/arduplane.param b/arduplane.param
index bab3732..0be2de1 100644
--- a/arduplane.param
+++ b/arduplane.param
@@ -31,7 +31,7 @@ ARSPD_AUTOCAL,0
 ARSPD_BUS,1
 ARSPD_FBW_MAX,30
 ARSPD_FBW_MIN,20
-ARSPD_OFFSET,159.8796
+ARSPD_OFFSET,162.0919
 ARSPD_OPTIONS,0
 ARSPD_PIN,65
 ARSPD_PRIMARY,0
@@ -167,8 +167,8 @@ COMPASS_CAL_FIT,16
 COMPASS_CUS_PIT,0
 COMPASS_CUS_ROLL,0
 COMPASS_CUS_YAW,0
-COMPASS_DEC,0.1082827
-COMPASS_DEV_ID,0
+COMPASS_DEC,0.108115
+COMPASS_DEV_ID,81923
 COMPASS_DEV_ID2,592905
 COMPASS_DEV_ID3,0
 COMPASS_DEV_ID4,0
@@ -203,7 +203,7 @@ COMPASS_MOT3_Z,0
 COMPASS_MOTCT,0
 COMPASS_ODI_X,-0.01878012
 COMPASS_ODI_Y,-0.001530154
-COMPASS_ODI_Z,0.007259261
+COMPASS_ODI_Z,-0.004679582
 COMPASS_ODI2_X,0
 COMPASS_ODI2_Y,0
 COMPASS_ODI2_Z,0
@@ -214,9 +214,9 @@ COMPASS_OFFS_MAX,1800
 COMPASS_OFS_X,-154.3344
 COMPASS_OFS_Y,-277.8472
 COMPASS_OFS_Z,58.94371
-COMPASS_OFS2_X,101.903
-COMPASS_OFS2_Y,-18.47033
-COMPASS_OFS2_Z,-35.31593
+COMPASS_OFS2_X,99.2626
+COMPASS_OFS2_Y,27.90453
+COMPASS_OFS2_Z,-40.68139
 COMPASS_OFS3_X,-82.29945
 COMPASS_OFS3_Y,-249.0217
 COMPASS_OFS3_Z,84.73182
@@ -228,8 +228,8 @@ COMPASS_PMOT_EN,0
 COMPASS_PRIO1_ID,81923
 COMPASS_PRIO2_ID,592905
 COMPASS_PRIO3_ID,0
-COMPASS_SCALE,0.9780152
-COMPASS_SCALE2,1.183262
+COMPASS_SCALE,0.9875948
+COMPASS_SCALE2,1.152302
 COMPASS_SCALE3,0.9780152
 COMPASS_TYPEMASK,54207
 COMPASS_USE,1
@@ -340,8 +340,8 @@ FWD_BAT_VOLT_MIN,0
 GCS_PID_MASK,0
 GLIDE_SLOPE_MIN,15
 GLIDE_SLOPE_THR,5
-GND_ABS_PRESS,100729.4
-GND_ABS_PRESS2,100759.3
+GND_ABS_PRESS,101172.8
+GND_ABS_PRESS2,101269.5
 GND_ABS_PRESS3,0
 GND_ALT_OFFSET,0
 GND_EXT_BUS,-1
@@ -413,17 +413,17 @@ INS_FAST_SAMPLE,5
 INS_GYR_CAL,1
 INS_GYR_ID,2359586
 INS_GYR2_ID,2229282
-INS_GYR2OFFS_X,-0.03559214
-INS_GYR2OFFS_Y,-0.012333
-INS_GYR2OFFS_Z,0.0168079
+INS_GYR2OFFS_X,0.03318984
+INS_GYR2OFFS_Y,0.01502704
+INS_GYR2OFFS_Z,0.001204156
 INS_GYR3_ID,2360330
-INS_GYR3OFFS_X,0.005726049
-INS_GYR3OFFS_Y,0.004109862
-INS_GYR3OFFS_Z,-0.009601761
+INS_GYR3OFFS_X,-0.008034397
+INS_GYR3OFFS_Y,0.000338572
+INS_GYR3OFFS_Z,-0.006852113
 INS_GYRO_FILTER,20
-INS_GYROFFS_X,0.003651475
-INS_GYROFFS_Y,-0.00555335
-INS_GYROFFS_Z,0.0002026637
+INS_GYROFFS_X,0.005511933
+INS_GYROFFS_Y,-0.01647778
+INS_GYROFFS_Z,0.001096812
 INS_HNTCH_ENABLE,0
 INS_LOG_BAT_CNT,1024
 INS_LOG_BAT_LGCT,32
@@ -1167,10 +1167,10 @@ SR3_RAW_SENS,1
 SR3_RC_CHAN,1
 STAB_PITCH_DOWN,2
 STALL_PREVENTION,0
-STAT_BOOTCNT,62
+STAT_BOOTCNT,71
 STAT_FLTTIME,1903
 STAT_RESET,165497600
-STAT_RUNTIME,83922
+STAT_RUNTIME,90340
 STEER2SRV_D,0.005
 STEER2SRV_DRTFCT,10
 STEER2SRV_DRTMIN,4500
@@ -1182,7 +1182,7 @@ STEER2SRV_MINSPD,1
 STEER2SRV_P,1.8
 STEER2SRV_TCONST,0.75
 STICK_MIXING,0
-SYS_NUM_RESETS,97
+SYS_NUM_RESETS,106
 SYSID_ENFORCE,0
 SYSID_MYGCS,255
 SYSID_THISMAV,1
```
Nobody from the team remember anything about fiddling with compasses during this time and yet SCALE has changed, the same as some OFS and ODI. Also DEV_IDs are now set as I originally wanted. This should be as in ```Then after compass calibration (via MissionPlanner):``` form post above. I don't know what has happened.