ArduPilot/ardupilot #18325: Wrongly initialised structure will compile

teoPiD, 2021-08-13:
## Bug report

This was discussed on discord general thread. I was advised of opening an issue (not quite sure if this is a bug or was a design choise)

**Issue details**

It seems that waf will compile a wrongly initialised structure (which can be tested in the following branch https://github.com/teoPiD/ardupilot/tree/test-compilation).  https://www.jdoodle.com/online-compiler-c++/, http://cpp.sh/ and https://www.onlinegdb.com/online_c++_compiler throw an error if a structure like the following one is compiled, while waf doesn't:

```
void send(uint8_t id, uint8_t len, uint8_t reg, uint16_t value)
{

    struct packet {
        uint8_t id = id;             //#1 Servo ID
        uint8_t length = len + 3;         //#2 Packet length 
        uint8_t instruction = INST_WRITE;    //#3 Instruction is a Write
        uint8_t reg;            //#4 First parameter is the register
        uint16_t value;         //#5 Following parameters is the value
    } msg;
```

Compilation of this structure will lead to an error in behaviour, since id and length will not hold the supposed values.

Error in  jdoodle, using GCC-9.1.0 (but also thrown with older versions of GCC, including 4.9.2, 5.1 and 7.2)
[image]

**Version**
Copter-4.0.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Compile for Pixhawk1, haven't tested other hardware/platform/version of code

peterbarker, 2021-08-14:
@teoPiD when I pasted this into  master's `ArduPlane.cpp` and compiled on CubeBlack with
```
pbarker@bluebottle:~/rc/ardupilot(master)$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I got this:

```
../../ArduPlane/ArduPlane.cpp: In function 'void send(uint8_t, uint8_t, uint8_t, uint16_t)':
../../ArduPlane/ArduPlane.cpp:315:6: warning: no previous declaration for 'void send(uint8_t, uint8_t, uint8_t, uint16_t)' [-Wmissing-declarations]
 void send(uint8_t id, uint8_t len, uint8_t reg, uint16_t value)
      ^~~~
../../ArduPlane/ArduPlane.cpp:320:26: error: use of parameter from containing function
         uint8_t length = len + 3;         //#2 Packet length
                          ^~~
compilation terminated due to -Wfatal-errors.
```

So I think we're OK - that's been our release compiler for donkey's years - 'though we've changed now.

Note that we'd probably go with this:

#include <stdint.h>

void send(uint8_t id, uint8_t len, uint8_t reg, uint16_t value)
{

```
    const struct packet {
        uint8_t id;             //#1 Servo ID
        uint8_t length;         //#2 Packet length 
        uint8_t instruction;    //#3 Instruction is a Write
        uint8_t reg;            //#4 First parameter is the register
        uint16_t value;         //#5 Following parameters is the value
    } msg {
       id: id,
       length: uint8_t(len +3),
       instruction: INST_WRITE,
       reg: reg,
       value : value
    };
}
```

There are other issues in play here - like the packet length being inferred and the fact this should proabable be caled `send_set_register` :-)

And it might be able to be const, too :-)

Should we close this as "we don't have this problem and our compiler detects it?"  There's the question as to why it doesn't on your machine, I guess?

teoPiD, 2021-08-14:
@peterbarker First of all, thanks for your comment. The struct/function was just an example, in reality it isn't implemented like that and I think they are according to your inputs (was to show that it compiled).

I opened this issue to understand why it does compile without an error, which goes according to your question: "There's the question as to why it doesn't on your machine, I guess?" I have just compiled master, with this modification and it **DOES** compile **WITHOUT** an error. Do you think is just from my machine setup? It was done according to https://ardupilot.org/dev/docs/building-setup-linux.html#building-setup-linux

peterbarker, 2021-08-17:
@teoPiD which version of linux?

teoPiD, 2021-08-17:
> @teoPiD which version of linux?

@peterbarker Ubuntu 18.04 (I guess Linux kernel version 4.15)

peterbarker, 2021-08-18:
@teoPiD that happens to be the box I tested on here.

I can't reproduce with any of these:
```
#export PATH=$HOME/gcc/gcc-arm-none-eabi-5_4-2016q3/bin:$PATH
#export PATH=$HOME/gcc/gcc-arm-none-eabi-6_2-2016q4/bin:$PATH
#export PATH=$HOME/gcc/gcc-arm-none-eabi-6-2017-q2-update/bin:$PATH
#export PATH=$HOME/gcc/gcc-arm-none-eabi-9-2019-q4-major/bin:$PATH
#export PATH=$HOME/gcc/gcc-arm-none-eabi-10.3-2021.07/bin:$PATH
```

Which compiler are you using locally that doesn't error off?

teoPiD, 2021-08-18:
@peterbarker 
´´´
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/arm-none-eabi-gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Checking for program 'make'              : /usr/bin/make 
Checking for program 'arm-none-eabi-objcopy' : /opt/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-objcopy
´´´

peterbarker, 2021-08-19:
On Wed, 18 Aug 2021, teoPiD wrote:

> @peterbarker
> ´´´
> Using toolchain : arm-none-eabi
> Checking for 'g++' (C++ compiler) : /usr/lib/ccache/arm-none-eabi-g++
> Checking for 'gcc' (C compiler) : /usr/lib/ccache/arm-none-eabi-gcc
> Checking [...]

Weird.  Pretty sure that one works for me.

Suggest installing our current release compiler (available for download 
from firmware.ardupilot.org) and moving on :-)

teoPiD, 2021-08-19:
@peterbarker strange indeed.

Will close dispute then.

closed 2021-08-19 by teoPiD